Photonsters/anycubic-photon-docs #4: Does a comparison fit here

NardJ, 2018-07-03:
I am wondering if a comparison between the photon and other similar priced printers has a place here?

http://www.pandarve.link/DLPSLAComparison/Comparison.html

X3msnake, 2018-07-03:
Yes this would be a good info to have.

We might need to work on the info a bit, it needs to be more consistente across printers to be a good comparison. but you can submit the HTML to the git.

What folder would be the best for this type of info?

NardJ, 2018-07-03:
Maybe a new folder named something like a "Buying guide"? This could also include accessories to buy along with your Photon?

NardJ, 2018-07-09:
@X3msnake Added a new folder Comparison and placed the files there. Not sure how to change the folder name, will look into that.

X3msnake, 2018-07-09:
PDF or SVG

NardJ, 2018-07-09:
Added the PDF

NardJ, 2018-07-15:
Reformatted to 1 page

X3msnake, 2018-09-10:
i have merged this with the master branch ;)

closed 2018-09-10 by X3msnake