Photonsters/anycubic-photon-docs #7: Consider splitting up the "cool models for testing" section

moppius, 2018-08-20:
It would probably be more useful to split the Readme's "cool models for testing" section into two sections:
### 1. **Calibration Models**
This section would just have one or two standard calibration models that everyone tends to use for testing, and the second section would contain a list of links to sites like Thingiverse, CGTrader, TurboSquid and any other repositories of free or paid 3D-printable models.

### 2. Model Repositories
This would probably be of more general use than the current list which is a mix of calibration models, standard test pieces and just random other sculpts.

Everyone has a different definition of "cool" anyway 😄

moppius, 2018-08-20:
I'm quite happy to do this on my fork and submit a PR for the changes.

X3msnake, 2018-08-21:
i can add you to the photonsters team and u can do it directly on the
project repo.

X3msnake, 2018-09-10:
sent u an invite to the team

closed 2018-10-11 by moppius