Photonsters/anycubic-photon-docs #8: Safety of Resins

NardJ, 2018-09-08:
I just sent the following question to MonoCure about the safety of their resin. The answer will be input for the safety page.

> 
> Hi, I am trying to read your datasheet but it keeps me wondering if it is save to use your resin inside my home. 
> 
> I use it in an attic room (20 Celsius) inside a custom (laminar air flow) print enclosure which vents to the outside. Some odour does leak in the room when printing. (When opening the enclosure to remove the print and cleanup, I use a mask. The mask is a Skandia 1047454 Dual Half Mask with A1 filters. For cleanup I use ResinAway.)
> When not printing, I keep the resin in the printing vat (in the enclosure) with an almost airtight lid laid on top. 
> 
> I, my wife and son sleep one floor lower. Should I be worried?

X3msnake, 2018-09-08:
ok

NardJ, 2018-09-09:
So I got this back:
> We have been developing and manufacturing UV products for over 30 years! One of my Chemists has been with us the whole time. His health has not been affected by any of these chemicals. I have never seen him wear a mask or gloves! If our resin had any dangerous levels of VOC emission then the SDS would state that as a respiratory warning. It sounds like you are doing everything in the way of protection so all will be fine.

webbyeasy, 2018-09-09:
Thats what was said to me when I asked about it lol. But, Australian health
compliances are very strict, so I do trust the information in the SDS. But
in saying that SDS generally only applies to the product in its packaged
state and NOT by products of a reaction such as UV exposure. Just take the
necessary precautions such as active venting and making sure you dont cross
contaminate resin into the house.

NardJ, 2018-09-09:
Yeah the SDS  has some interesting snippets. I asked a follow up question. Curious what they will answer:

> I do have a follow up question and maybe I am interpreting the SDS wrong, but I found the following snippets in the SDS:
> "Avoid breathing vapours..."
> "Store in tightly closed containers in a properly vented storage area..."
> "Evidence shows, or practical experience predits, that the material produces irritation of the respiratory system, in a substantial number of individuals, following inhalation"
> "Long-term exposure to respiratory irritants may result in disease of the airways involving difficult breathing and related systemic problems."
> 
> I can off course imagine several reasons these are nothing to worry about, maybe this is not applicable due to small scale usage or only a problem when hovering directly above a resin pool. Can you tell me why this is in the SDS and what to avoid?

webbyeasy, 2018-09-10:
> @webbyeasy <https://github.com/webbyeasy> What kind of mask do you use
> btw? My current write down of safety is at https://github.com/
> Photonsters/anycubic-photon-docs/tree/Safety/safety. It is quite rough,
> but If you have suggestions/correction let me know!

its a good article, but why single out Monocure? I would be far more
concerned of resins manufactured in countries with far less material safety
laws then Australia.

I would put links to the SDS of most UV Resins in the article first and put
a caveat in if you plan to write only about Monocure. or just remove a
direct reference to Monocure and make the article fair and objective.

I know you are been objective, but reading it for the first time felt like
you are singling out Monocure. Lets not do that on our public docs.

When I've been writing the Phontonsters documentation Ive avoided referring
to particular products and company.

Also if you plan to quote Monocure in the docs from your email
correspondence, please make sure you have permission from them to do so
first.

PS: I use a painters mask

X3msnake, 2018-09-10:
> Thx for your feedback. I want to show that the resins don't have to be
> very dangerous, but on the other hand some can be very bad for your health.
> Maybe I should not mention manufacturers but only use snippets from various
> SDS's to show how to read them.
>
> I am not planning to quote Monocure from their mails. I cannot imagine
> they would allow this btw.
>
> I got the following reply back:
>
> This would be stated on any chemical based SDS. They have to cover
> themselves the best they can. Many household chemicals, such as bleach and
> orange cleaner are much more nasty then our resin! But the law states as
> since it goes directly to the public no SDS is required! The only reason we
> need an SDS is because of air travel requires it for any liquids.
>
> So we have an SDS on one hand which warns about the dangers as obligated
> by law and a sales person who tells me not to worry.

A good guy to talk about the resins is Stefan Guiducci on the Anycubic
photon Group

10/09/2018 à(s) 14:58:

jdlizard, 2019-01-18:
I think you can't totally trust a sds sheet to keep you safe. Research is limited right now on the resins and being over protective will never be a bad idea.  I work with these resins at work and we use vacuum hoods.  Not all of our resin printers are hooked to ventilation but all post processing is done under a hood. We do not wear respirators but we also never smell resin under the hood.

zeelobby, 2019-04-11:
Let's be honest. Their response is pretty indicative of how SDS are handled. It's basically a legal catch-all in case something happens. Better to put more and not worry when respiratory problems pop up 10 years from now. "Well we said watch out for vapors".