Photonsters/anycubic-photon-docs #10: IWantToHelp

bartwaw, 2018-10-22:
Hi,
here's more detailed description of chitu gcode settings - get them from Orbeat D100 distributor.
As i can see, D100 is using same board as Photon.
I've also connected external stepper driver - TMC2208. On board there are exposed soldering points for connecting external drivers.
Now i'ts silent - i suppose that on board is A4988.

Got also question - photon file editor.
can you make few improvements:
1. better file opening - that open dialog is annoying - is possible to make something similar to windows open dialog ?
2. can you add file types .cbddlp

cheers
bartek

heres what i've got from distributor.
Hope it helps

-------------------------
;[[After the semicolon is a comment, please modify the corresponding parameters according to the comment, the command after the semicolon will be ignored]]
The basic format is Mxxx Ixxx, Mxxx Txxx or Mxxx Sxxx, where Ixx (which is the letter I, the first letter of the integer, not the number 1) followed by an integer, which can be either decimal or hexadecimal (starting with 0x) ), followed by Sxx is a floating point number

;

M8513

[Stepper motor direction control] The direction of I1 and I-1 is just the opposite, so if the motor direction is wrong, either change the wiring or change the direction parameter.
M8002 I-1; X stepper motor direction, I1 or I-1
M8003 I1 ; Y stepper motor direction
M8004 I-1 ; Z stepper motor direction

;[XYZ axis extrusion head/platform movement configuration] only affects the actual movement direction of the manual interface button;;;;;;;;;;;;;;;;;;;;;;;;;;
M8005 Z0 ; 0: Z-axis direction, forming bracket movement,
; 1: Z-axis direction resin groove movement

;
; [Speed / Acceleration Settings] Speed in mm / s, acceleration in mm / s ^ 2
M8006 I40; the maximum starting speed, when the speed of movement has passed this speed, it will start at this speed.
This speed is mainly to prevent out of step, this value is too small, will cause the rail angle bump when rounding
M8007 I25; the maximum rail bending speed value (corresponding to the jerk speed in the open source firmware), if the actual rail bending speed of the motion is greater than this value,
; will force the movement to slow down. When filling, there will be a lot of reciprocating motions, this value is large, the noise is large, and it is easy to lose steps.
; This value is small, slow, and print quality is relatively poor when printing speed is high
M8008 I1000; Acceleration, the larger the value, the larger the average speed of actual operation, but the noise is also large, the value is small, and the actual speed will be smaller.
;
;[[[step related parameters]]] After the parameters are set, please print a cube, then measure the size with a ruler to confirm that there is no problem with the parameters.
;
M8009 S0.0125; [XY value of each step of XY] is 20 teeth, the pitch is 2mm (mxl specification, 2gt is 2mm), 1.8 degree stepping motor,
; 16 subdivision, then (20 * 2) / ((360 / 1.8) * 16)
;M8009 X0.0125 Y0.0125 ; If you need to set the X, Y stepper motor parameters independently, you can use this command
;
M8010 S0.000625 ; [Z mm value per step] On a non-delta machine, calculate the formula: lead / ((360 / 1.8) * 16).

;
;[Various speeds maximum] In order to ensure the machine can be stable, please set according to the measured results.
M8012 I200; maximum speed of XY motion mm/s
M8013 I5 ; maximum speed of Z motion mm/s

;
;[zero speed]
M8015 I10 ; The first zero return speed when Z is zero, the speed is faster, the Z movement speed of the manual interface is also the same as this speed, and the Z axis is also used for this speed.
M8015 S30; Zero return speed when XY is zero, the speed is faster, the XY moving speed of the manual interface is also the speed of the XY axis shifting
M8016 I6 ; The second zero return speed when Z returns to zero, the speed is slower. Decreasing the second zero return speed can improve the repeat positioning accuracy of the limit. If the parameter is 0, the second zero is not performed.

[Z-axis demoulding rise and fall speed] During the demoulding process, first slowly rise, then rise quickly, then quickly fall
M8015 T2 ; Z-axis slow release rate
M8016 T6; Z-axis demoulding rapid rise and fall speed
;

; [When the Z axis is demolded, the dwell time after the rise]
M8016 D0; the unit is ms (milliseconds),
;
[... 91 lines cut ...]
;
;[Start LOGO time]
M8085 I3000; boot logo duration, minimum 100ms, maximum 6000ms

;[Screensaver time]
M8085 T0; How long does it take to enter the screen saver interface, the unit is seconds (s), when it is 0, it means that the screen saver is disabled.

;[External Drive Settings]
M8087 I0 T0 ; I: Settling time of the direction signal to the high level of the pulse signal, in ns
; T: The minimum hold time of the pulse signal, in ns. If there is no external drive control board, set it to zero.
; Several external drive parameters: THB7128: I100000 T0 ;;;; TB6560: I40000 T0 ;;; TB6600: I100000 T0
;
;
;
;[Action after printing is completed]
M8489 P3 ; action after printing is completed
; 0: Turn off all motors,
; 1: Does not work after printing,
;2: After printing is completed, return to zero, the motor is not powered
;3: After the printing is completed, it drops to the maximum stroke of Z, and the motor is continuously powered.

;
;This parameter must not be less, otherwise the parameter cannot be saved to the device.
M8500; save configuration
;

X3msnake, 2018-10-26:
@bartwaw 
Greetings Master, Photon File Editor requests should be adressed at issues on Photon File Editor Project.

Thanks for the file share and the tips on orbeat

X3msnake, 2018-11-04:
@bartwaw this orbeat file is the factory defaults for the orbeat?

can you try resin exposure finder on your orbeat?