Photonsters/anycubic-photon-docs #12: Measurements for the LED reflector

esseeayen, 2018-11-01:
I found this group and have a similar DLP printer, the wanhao d7, which we have been upgrading but someone mentioned that some of our upgrades may also be useful for the Photon. Is there any chance you can take some pictures and measurements of the space and holes for the reflector for the UV led?

Thanks!

esseeayen, 2018-11-03:
Hopefully! I that's the idea.

bartwaw, 2018-11-03:
i'm thinking about this also. But i will do it on orbeat printer.
Ordered such pcb 
https://www.aliexpress.com/item/free-shipping-1pce-12W-18W-20W-24W-30W-36W-48W-Rectangle-LED-Aluminum-Plate-High-Power/32722702900.html?spm=a2g0s.13010208.99999999.310.19203c00KrRbBi

which will cover whole LCD ( its 145x106 but i think that whole pcb dimension)

and 20 pcs 3W leds 405nm so they should give 60W.
But i will drive them little bit lower to get about 50w.

X3msnake, 2018-11-03:
beware that driving the leds at lower voltage may dift the led wavelenght.

what dimensions do you need?

Photon is much the same as orbeat, hardware wise so i would not be
surprised if the dimenstions are intermutable

esseeayen, 2018-11-04:
@bartwaw agreed, this is the reason why we were making one for the Wanhao D7, but we have done a lot of research and not just any "UV led" will do, even if it says 405nm.

@X3msnake i have attached the dimensions i need. It would be amazing if they were the same size 👍 

Thanks in advance for your help.

[image]
[image]

esseeayen, 2018-11-04:
This was our version 1 which didn't work well, but we're onto trying something new!

[image]

bartwaw, 2018-11-04:
esseeayen i see that you used 5050 led's.
I've read somewhere that isn't best option.
I think that main problem is that they are almost 180 degrees.
That's why i'm thinking about such leds

https://pl.aliexpress.com/store/product/High-Power-LED-Chip-50W-UV-365nm-50-Watt-Light-Beads-UVA-Purple-COB-For-polymer/2335145_32755950096.html?spm=a2g17.12010612.8148356.1.15c42989qzJgjU

best way is to find good leds with 60 degree beam like them:
https://www.aliexpress.com/item/10PCS-Imitation-cree-led-3535-UV-LED-365nm-370nm-380nm-385nm-395nm-400nm-405nm-3W-Purple/32893869506.html?spm=2114.search0104.3.1.62c52a70TXswZq&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_319_317_5728817_10696_10084_10083_5729417_10618_10304_10307_10820_10301_10821_5730517_5733317_5733217_10843_5733417_328_10059_10884_5732917_10887_100031_5733117_321_322_10103_5733517_5733617,searchweb201603_55,ppcSwitch_0&algo_expid=fc3b6d59-3471-4239-bab2-ecec3bd98c87-0&algo_pvid=fc3b6d59-3471-4239-bab2-ecec3bd98c87

but i didn't found aluminium PCB's that have dimensions over 120x60

anyway i'm thinking about simillar to this:
https://www.fepshop.com/shop/parts/led/paraled-3rd-party-kit/

esseeayen, 2018-11-04:
@bartwaw 5050 Leds beam spread wasn't the angle that was the issue. It's actually the lumens/watt that is the issue.

We have a solution now, but we are custom making the PCB and everything to make sure it works properly.

I know the people from FEP Shop and they said the paraled is good, but they seem to be no longer available as a DIY part, as well as being >$200!

bartwaw, 2018-11-04:
so what leds will you use ?

bartwaw, 2018-11-04:
any more :) what type

esseeayen, 2018-11-04:
you'll see :)
ᐧ

esseeayen, 2018-11-06:
If anyone is happy to measure, happy to provide a board for testing (free, but just pay the shipping)

closed 2019-05-15 by X3msnake

madeintaiwan, 2020-03-20:
@esseeayen Did you manage to produce upgraded LED units?