Photonsters/anycubic-photon-docs #13: IWantToHelp  - gcode for mixing resin in vat

bartwaw, 2018-11-04:
Want to share my Gcode that i'm using to mix resin in VAT.
It's useful specially for colour resins - i got wanhao grey resin water washable, after one day when it stays in resin vat dye stays on bottom.

X3msnake, 2018-11-04:
what do you mean?
gcode to mix resins in the photon?

how do you achieve that?

bartwaw, 2018-11-04:
Easy,
when resin stays in VAT for longer time, dye (or pigments) thanks to gravity will goes down.
This is gcode that moves z axis up and down in mixed lengths to mix resin.
It mixes resin by build platform :)
It really works on my Orbeat, which I said has same board as Photon. So i should work at 99% :)
Check it :)

X3msnake, 2018-11-04:
post it here in the issue and i will give it a try

X3msnake, 2018-11-04:
missed the zip. nevermind will test it

X3msnake, 2018-11-04:
mmm, that's plaing Gcode. not sure if it will work... testing now  ;)

X3msnake, 2018-11-04:
@bartwaw @Photonsters/photonsters 
https://youtu.be/M1JdoMIhdcM

looks like it does work. @bartwaw more important than the mixer, that is cool but needs to en above the endstop, where did u get the gcodes from?

do you have acess to a list of all available codes?

we can open up a page for this and respective community tool programs if you do. makes more sense that way actually.

and thanks for sharing this

X3msnake, 2018-11-04:
@bartwaw 

M17 /18 is as good as not being there, compleatly ignored by the photon at least

X3msnake, 2018-11-04:
G90/91 is also ignored, printer works only in absolute coordinates

X3msnake, 2018-11-09:
@bartwaw 

Can you test these codes?

[image]

rsilvers129, 2020-04-16:
Can you please create a mixer that will run much longer? Sometimes I run the mixer manually 10x and wish I could just have one that ran once for 15 minutes. If you think the motors will get hot, it is ok if it rests for 30-60 seconds every few minutes.

yy-cube, 2020-04-23:
@rsilvers129 

You can edit this gcode with a text editor. It is really easy. Just copy & paste those mix modes (there a different types, just watch your printer when hes mixing) as often as you want / need them and you are ready to mix.

berkakinci, 2020-12-29:
Has anyone put this in gcode in the repo?  (Sorry; I'm not familiar enough to suggest where.)
I'm happy I happened across this issue.  I was about to make my own gcode.

berkakinci, 2020-12-30:
I'm attaching my takes on the resin mixing.  I made some slow and long mixes; they work with my resin vat heater.
I'm hoping these will make it in the repo as well.

[Mix Resin.zip](https://github.com/Photonsters/anycubic-photon-docs/files/5755513/Mix.Resin.zip)