Photonsters/anycubic-photon-docs #20: have purchased this machine in china in 2019 from factory and got delivered in ghungzou and then got shippment locally to pakistan , i have tested it yesterday  , it is not producing 3d form , when i am printing the block it produce which is not solid , what should i do

k3g1397, 2019-04-15:
have purchased this machine in china in 2019 from factory and got delivered in ghungzou and then got shippment locally to pakistan , i have tested it yesterday  , it is not producing 3d form , when i am printing the block it produce which is not solid , what should i do

half block remains with upside and half on downside , it breaks while printing block

[image]
[image]
[image]
[image]
[image]

Sailormaine, 2019-04-15:
This is my theory - no guarantees, but it might help.

I think that the resin that Anycubic is providing with the machine has changed, but the demo file has not.  I think it needs a bit more exposure.

Start the demo file again.  As soon as you have started it, press the little gear icon on the bottom right of the touch screen.  Edit the bottom setting - I think it's called "normal exposure" - and raise the value by 2 to 4 seconds.  Then click the checkmark icon and then the return icon to bring you back to the regular screen.  

Let's see how well that works.

k3g1397, 2019-04-15:
Ok ill do it

X3msnake, 2019-05-15:
This type os questions should be placed ate fb group not here
https://www.facebook.com/groups/AnycubicPhoton/

closed 2019-05-15 by X3msnake