Photonsters/anycubic-photon-docs #32: The FAQ states to find the fan_off.gcode on the Facebook group, but doesn't specify *which* group

Taegost, 2019-11-06:
I'm trying to find a way to disable the fans on my Photon S without physically modifying it (yet), but I can't seem to find the fan_off.gcode mentioned in the FAQ.
Can someone include a link directly to it instead?  Or better yet, pull it into the repo so it can be version controlled and easily accessible?
Thank you.

X3msnake, 2020-02-04:
photon S does not have a equivalent tool like the defaults.gcode since it
is a different board

only removing it physically

Taegost, 2020-04-29:
> photon S does not have a equivalent tool like the defaults.gcode since it is a different board only removing it physically No dia quarta-feira, 6 de novembro de 2019, Mike Wheway < notifications@github.com> escreveu:
> […](#)
> -- Com os melhores cumprimentos, Vinicius Silva

I found as much in my research, but we still need to update the FAQ with the actual code or point to where it can be found, I was never able to find it at all, and I've tried finding it for other users on Reddit that have the original Photon.

X3msnake, 2020-04-29:
gcode modifications only apply to the Photon Classic and CBD-Tech
motherboards

there  is no way to do this that we know of. contact Anycubic about it and
if you find out do share back

à(s) 14:29:

X3msnake, 2020-04-30:
@Taegost again, this will only work for CBD-TECH Boards > Photon Classic / Mars Classic / Orbeat / Epax X1 / Qidi Shadow / ...

[image]

https://github.com/Photonsters/anycubic-photon-docs/blob/master/firmware/factory-defaults-translated.gcode

X3msnake, 2020-04-30:
@Taegost Also the correct approach is to dump your printer config file istead of using the archives one since settings might vary from vendor to vendor

Open a txt file and drop these lines, change the extension to .gcode and print it in the printer (only Chitubox cbd-tech boards)

```
;ENABLE M8093 TO DUMP DEBUG TO FILE - Only works in recent firmwares
M8093 I1 
```

closed 2020-04-30 by X3msnake