Photonsters/anycubic-photon-docs #36: new printer killed 3 Usb-Sticks

GreBra, 2020-01-08:
Hello photonsters,
I bought the photon a few days ago and i am new in the field of 3D-printing. The setup went well. However sometimes the touchscreen flickers and the included USB-Stick, as well as two of my own, were killed within minutes. At least i managed to print 1% of the photon.photon until my last USB-Stick was destroyed.
Is there any known issue or possibility to fix this easily or should I send it back (I don't want to risk my warranty)?
Thanks a lot for helping!

X3msnake, 2020-01-09:
send it back if you still can

no know issues like you describe.

bain2236, 2020-01-26:
My photon has also killed the included usb and another usb of mine.

Attempting with a new usb, if this fails it's going back, providing I am still able to send it back