Photonsters/anycubic-photon-docs #37: Non-standard use of the printer

dark184, 2020-01-09:
Hello. I have a printer, ANYCUBIC PHOTON. I want to use it for more than just printing. Tell me if I can display an arbitrary image without using slicer (preferred image format *.BMP, or any other)? And turn on the lights for a few seconds?

X3msnake, 2020-01-09:
use photon file editor

dark184, 2020-01-09:
I found exactly what I need here
https://github.com/Reonarudo/pcb2photon
This is a Gerber to file Converter *.photon
But to my great regret it all works under Linux. Maybe there is a similar utility for WINDOWS?

X3msnake, 2020-01-09:
use photon file editor from photonsters.org.
there are other project that do similar things listed under community
projects

dark184, 2020-01-09:
Thanks. Installed file editor, prepared a picture in PNG format 1440x2560. Trying to upload to "file editor"... Only black screen. Am I doing something wrong?
[image]
[image]