Photonsters/anycubic-photon-docs #39: Updating from 4.2.17 to 4.2.18/9 causes LCD errors.

Shir0n1379, 2020-01-22:
When I perform a UV test on 4.2.17 my Photon is fine. However, if I then update the firmware to 4.2.18/9 (Anycubic or Epax) the LCD begins to flicker whenever the UV test is performed.

X3msnake, 2020-01-22:
you have a bad solder joint on the Dram.
dessoldering cleaning up and ressoldering with normal leaded solder is a
permanent fix

Shir0n1379, 2020-01-22:
Why does this only happen when I upgrade to 4.2.18 and fixes when I reflash to 4.2.17?

X3msnake, 2020-01-22:
Just because you dont get it with .17 does not mean it is not there.
You will end up getting frame shearing with .17 sooner or later, as soon as
you slice a model with enough checkered details to overflow the memory
buffer.

only safe firmwares are 4.0.11 (no AA) and 4.2.18 (Epax) all other
firmwares have issues

A way to test for this is filling a plate with the calibration matrix
model, slicing it and running a dry run test.

à(s) 09:41:

Shir0n1379, 2020-01-22:
Okay thanks for the help, are there any videos on how to clean and resolder the Dram available?

X3msnake, 2020-01-22:
Not that I am aware of.

You need a hot air station and flux.
You first desollder the chip, and clean the contacts with good amounts of
flux, braided copper and a 40w solder iron with a flat tip
You can then proceed to solder the chip with a microsoldering tip and some
leaded solder

Similar to this video but you can use solder wire instead of paste like
they do
https://www.youtube.com/watch?v=E3OmchO_mDc

If you are not experienced in doing SMD soldering i recommend you go to
your local hackerspace or electronics guru and ask for help.

You can also try fw version 4.0.11 from photonsters fw archive. It does not
have any AA but it is a solid firmware

à(s) 12:45:

Shir0n1379, 2020-01-22:
Okay not sure if this changes anything but just flashed my firmware to 4.0.11 and the same error occured

Shir0n1379, 2020-01-22:
If it helps this is the problem: https://youtu.be/uTGT1G09x-g

X3msnake, 2020-01-22:
mmm, that looks different than what i was picturing

no harm in doing the SDRAM resoldering since 90% of the graphic issues on
these boards are caused by that, but that but to be honest that is the
first time i see this issue pop up on a firmware upgrade/downgrade, usually
this show up when a screen has been damaged by pressure near the contacts
or excess heat

à(s) 13:04:

Shir0n1379, 2020-01-22:
Yeh that was my original thinking when I updated I must have damaged the screen, but downgrading fixed it. Only thing I can think of is that the firmware that introduced frame-shearing actually fixed the memory overflow on my board by accident, but when frame-shearing was fixed my board started overflowing. As this also happens when I flash to the only other firmware that does not have frame shearing

Athens428, 2020-04-05:
Did you ever figure out what is causing this? I am having the exact same problem.  My screen looks just like yours after updating.

Fusseldieb, 2024-02-21:
I don't think that's a bad DRAM issue. Mine also does just this after I've updated it to 4.2.19. The display flickers now. Guess I'm going back to 4.2.17.