Photonsters/anycubic-photon-docs #40: EPAX X1 owners can update firmware to version 4.2.23

stef-ladefense, 2020-01-23:
hi,

this version is ok for original Photon ?

https://epax3d.com/pages/firmware

thx

SB

X3msnake, 2020-01-23:
yes
firmware 4.2.18 for epax X1 is currently the recommended firmware for the
photon classic

stef-ladefense, 2020-01-23:
and version 4.2.23 ?

X3msnake, 2020-01-23:
no reports on .23 yet

Rhiker1, 2020-02-26:
I upgraded to .23 today and it worked fine so far.  One print.

m0n5t3r0u5, 2020-07-07:
> I upgraded to .23 today and it worked fine so far. One print.

Any new insights so far about the firmware?

stef-ladefense, 2020-07-07:
personally i have this firmware since january and no printing problems

m0n5t3r0u5, 2020-07-08:
Thank you . I'll try it out.

m0n5t3r0u5, 2020-07-08:
By the way is aa active in this firmware?

Rüstem Taylan Wittig <r.taylan.akkus@gmail.com> schrieb am Mi., 8. Juli
2020, 07:31: