Photonsters/anycubic-photon-docs #45: LCD screen not functioning

lcanino, 2020-05-23:
Hi,

So I own an anycubic photon 3D printer. My screen star print my models with hole thought them and when I did a test the screen showed spots with black holes. I was advised to get a new screen do to the screen failing. So I went and got two ELEGOO 5.5 Inch 2K LCD. When I plugged it in it gave me two error codes T_100 and L_43. I do not know what L_43 is but I do know what the T_100. I then went and tried updating the firmware and to see if that help when that did not help I tried plugging in my back up that I got. When I installed I tried a screen test and it gave me a blank light square with no black square in the middle. I then tried to print something and got no error codes but nothing printed. If anyone has any idea how to fix this?
Any help in solving this problem would be grateful.

 Thank you in advance

X3msnake, 2020-05-25:
did you use the adapter that came with the screen and used it in the
correct position or have you simply tried to plug the straight cable to the
board?

X3msnake, 2020-05-25:
https://www.facebook.com/Photonsters/videos/1149686608729588/

Vinicius Silva <x3msnake@gmail.com> escreveu no dia segunda, 25/05/2020
à(s) 15:50:

> did you use the adapter that came with the screen and used it in the
> correct position or have you simply tried to plug the straight cable to the
> board?
>
> à(s) 05:21:
>
>> Hi,
>>
>> So I own an anycubic photon 3D printer. My screen star print my models
>> with hole thought them and when I did a test the screen showed spots with
>> black holes. I was advised to get a new screen do to the screen failing. So
>> I went and got two ELEGOO 5.5 Inch 2K LCD. When I plugged it in it gave me
>> two error codes T_100 and L_43. I do not know what L_43 is but I do know
>> what the T_100. I then went and tried updating the firmware and to see if
>> that help when that did not help I tried plugging in my back up that I got.
>> When I installed I tried a screen test and it gave me a blank light square
>> with no black square in the middle. I then tried to print something and got
>> no error codes but nothing printed. If anyone has any idea how to fix this?
>> Any help in solving this problem would be grateful.
>>
>> Thank you in advance
>>
>> <https://github.com/Photonsters/anycubic-photon-docs/issues/45>, or
>> unsubscribe
>>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

lcanino, 2020-05-25:
> did you use the adapter that came with the screen and used it in the correct position or have you simply tried to plug the straight cable to the board? lcanino <notifications@github.com> escreveu no dia sábado, 23/05/2020 à(s) 05:21:
> […](#)
> -- Com os melhores cumprimentos, Vinicius Silva

Yes I used

lcanino, 2020-05-25:
> did you use the adapter that came with the screen and used it in the correct position or have you simply tried to plug the straight cable to the board? lcanino <notifications@github.com> escreveu no dia sábado, 23/05/2020 à(s) 05:21:
> […](#)

Yes I used the adapter and wore gloves when installing it.

closed 2020-05-25 by lcanino