Photonsters/anycubic-photon-docs #46: FAQ request - curing

dcuccia, 2020-06-03:
FWIW, I read through the FAQ and did not see any info on post-print curing boxes. Would like to learn about common OTS and DIY options, including info on LED wavelength and power. Thanks!

X3msnake, 2020-06-16:
check the main page under community projects

X3msnake, 2020-06-16:
As for wavelenghts you should stick to 405nm
Old fluorescent Nail Spas have a 395nm center peak but work as well albeit
a bit slower yet can cause transparent resins to yellow

Vinicius Silva <x3msnake@gmail.com> escreveu no dia terça, 16/06/2020 à(s)
11:37:

> check the main page under community projects
>
> 03:51:
>
>> FWIW, I read through the FAQ and did not see any info on post-print
>> curing boxes. Would like to learn about common OTS and DIY options,
>> including info on LED wavelength and power. Thanks!
>>
>> <https://github.com/Photonsters/anycubic-photon-docs/issues/46>, or
>> unsubscribe
>>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva