Photonsters/anycubic-photon-docs #47: Off-time

Teddyz, 2020-07-10:
I just upgraded my Photon to 4.2.18 SW.

When printing I now see that it moves the platform, exposes and then has the off-time. 
I was expecting, move platform, off-time and then expose.

Is it just me or is it for everyone?

I made the .photon using original slicer, then editing it and changed exposure time in Photon File Validator. Now printing...

Teddyz, 2020-07-10:
[VID_20200710_202246.zip](https://github.com/Photonsters/anycubic-photon-docs/files/4904767/VID_20200710_202246.zip)
[Ant_MM_200um_t8_o6_be60_bl3-2.zip](https://github.com/Photonsters/anycubic-photon-docs/files/4904768/Ant_MM_200um_t8_o6_be60_bl3-2.zip)

Added a film and the file I am printing, if it can help. 

I was reading about off-time here: https://all3dp.com/2/anycubic-photon-slicer-all-you-need-to-know/

X3msnake, 2020-07-11:
off time always happens after the peel move it is used to add time to let
resin flow before exposing

Teddyz, 2020-07-11:
Yes, that is what I want to happen.  But it does not. FW version seems to not matter. I've tested 4.2.14, 18, 19 and all seems to behave similar in this respect.

I would be glad if someone would want to test this file: [off-time_tests.zip](https://github.com/Photonsters/anycubic-photon-docs/files/4907154/off-time_tests.zip)
2plugg+2hylsa_005mm_t2_o25_be2_bl2-2.photon
Exposure time: 2s
Off time: 25s
Bottom-layer-time: 2s
Number of bottom layers: 2

On my printer, it behvaves OK for first layer.
Second layer and on: It moves platform, it waits about 16s, expose 2s, waits some 3 seconds.
Expected: Move platform, wait25s, expose 2s, wait no (or little) time.

The second file in the zip is the same, but has off-time set to 5s. It seems to wait no time after moving the platform.

Thanks for helping!
[off-time_tests.zip](https://github.com/Photonsters/anycubic-photon-docs/files/4907154/off-time_tests.zip)

X3msnake, 2020-07-12:
bottom will be considered fulfilled if the peel time is above the off time.

you can calculate what a peel time takes as value by multiplying the travel
distance by the time units

closed 2020-10-15 by X3msnake