Photonsters/anycubic-photon-docs #53: Is there a way to level without touchscreen?

TheTh1rt33nth, 2020-10-29:
Hello to photon users! Is there a way to home photons without touchscreen? Maybe via usb or so
My touch screen is not working in right-down corner, so i can not press the "back" button and therefore level the bed. I hope there is an alternative way to do this

X3msnake, 2020-10-29:
wierd. touchscreen fail mode is not usually like that. try using a allen key to gently press the back button.

there is a number of actions you can perform via universal network sender, like uploading and managing prints and print files. and you can also do that via serial

if you mean levelling as in setting zero offset. that command however is not available afik

but if you mean home, the command can be sent via terminal with a G28 Z

if you have a Esp8266 you can install it permanantly inside your case with esplink firmware and use the terminal to send command accessing it via your web browser

X3msnake, 2020-10-29:
list of commands can be found on the blueprint folder of this documetation project