Photonsters/anycubic-photon-docs #58: USB read issues

Guthur, 2021-05-09:
Hi

I just wanted to share my theory on the common USB read errors with some of the AnyCubic models. 

I had issues where it would not read Lexar USB sticks at all and from what I gathered AnyCubic recommend using <8GB sticks, mine was 16GB.

But I don't actually believe it's the capacity itself as I have now got a SanDisk 16GB to work.

The reason I selected the SanDisk was due to having read an number of reports that the +5V line on the USB interface is sometimes relatively low and so I started to believe that there maybe a power delivery issue and SanDisk appears to have relatively low power consumption.

I think this is why <8GB is recommended and may often times work more reliably as it's likely got less power draw on average.

I'm not sure if we could figure out some mod to provide more reliable power delivery to the USB, or potentially use a USB drive that has external power supply (I've not tried this and it would be less convenient than a flash drive anyway).

X3msnake, 2021-05-09:
did you measure the power supplied to the USB and confirmed this to be a
issue?

Guthur, 2021-05-09:
@X3msnake I've not got the equipment to do so at the moment, I'm basing a number individuals who have measured it.

I'll try to get hold of something to measure it.

X3msnake, 2021-05-10:
Would be interesting to know if this is the case.

ChronoDK, 2021-12-19:
I got my USB working by formatting with this tool: https://rufus.ie/en/
Select non-bootable, MBR, FAT32.

mattwithoos, 2022-05-18:
**TL;DR: This is extremely unlikely to be the cause, see @ChronoDK 's comment.**

Stumbled across this by accident. I have a faulty Photon Mono X and can (perhaps) rule this theory out, at least for the X.

As the USB spec tolerates +-5% under load, a problematic "low voltage" would be anything under 4.75v. I note a stable 4.98v at rest and under load, but respect that others may have different readings for different models and under different loads for different motherboards.

Honestly, given @Guthur doesn't mention what formatting was factory-applied to the Lexar thumbdrive, I theorise that Lexar drives come preformatted to GPT, which is obviously incompatible - the docs advise formatting to MBR. Tangential evidence online suggests Lexar thumbdrives come preformatted as GPT - https://superuser.com/questions/1438615/usb-drive-not-working

Further, the power draw of a thumb drive is not worth even thinking about. We're talking tiny milliamp ranges, with a max (afaik) of 100mA before you need to write complex firmware logic to request more voltage from the host. You'd likely see other major issues elsewhere if power delivery was so poorly designed that they failed on the USB spec. It's just really unlikely and an unfair assumption to cast doubt over the circuit designers' efforts.

While I don't think I need to say more, the original post contains a lot of hearsay, so I'll share this: https://www.researchgate.net/figure/Power-consumption-of-all-devices-idle-read-write-under-USB-11_fig1_4361182 - this not only categorically rules out that the power consumption is a concern, but also completely refutes @Guthur's key assertions on SanDisk's alleged quality. Namely:
a) SanDisk are NOT more efficient (open to being corrected, can you link where you read this?)
b) SanDisk are actually the **worst** offenders for power consumption

In fact, as their storage goes up, they use more power! So unless the theory is that more power consumption by the USB supported by a high-quality power delivery system is the culprit (which would make it an oxymoron), I think we can rule out the USB port.

I think the nail in the coffin is @ChronoDK 's confirmation that USB read issues were resolved by formatting it as per the docs, which is also the standard advice throughout the community, and also advice which seems to work.

It may make sense to close this issue to not confuse future readers, unless @Guthur is happy to share:
 * A link to the power draw of a SanDisk USB, and a couple of comparable comparison USB powerdraws
 * A link to the discussion where all these "individuals who have measured it" shared their results